EnotionZ/gpio #35: Cant read initial value of pin

nursultan156, 2015-02-18:
Hi,

i encountered with the problem of reading initial value of pin inside ready function, here is the code snippet

var gpio11 = gpio.export(17, {
    direction: "in",
    interval: 200,
    ready: function (val) {
        //here, how to read value of pin?
        console.log(val); // "undefined" - always
	console.log(gpio11.value); // "0" - always
		
        gpio11.on("change", function (val) {
            console.log(val); // reads correctly, "1" or "0"
        });
    }
});

rzr, 2018-06-28:
According to:

https://github.com/EnotionZ/GpiO/blob/master/lib/gpio.js#L114

ready has no parameters, I wish It has a status argument to trap errors

rzr, 2018-07-04:
I have fixed this issue please confirm, it may relate to related bug:

https://github.com/EnotionZ/GpiO/issues/34

https://github.com/EnotionZ/GpiO/issues/38

EnotionZ, 2018-10-01:
The patch was already merged into master. Tests are in #58 and will be merged into v1 branch. Closing issue.

closed 2018-10-01 by EnotionZ