EnotionZ/gpio #38: How do you initialize a pin to high?

jfarago, 2016-01-31:
I can initialize the pin and then set it high in the ready callback but for a split second it is pulled low.

cagodoy, 2017-03-08:
i think you should be use ready function.

```
const gpio4 = gpio.export(4, {
  direction: 'out',
  ready: () => {
    // sets pin to high in the first time
    gpio4.set()
  }
})
```

jfarago, 2017-03-08:
@cagodoy yes, that is what i ended up doing but its a poor solution as it toggles the gpio on restart.

cinderblock, 2017-11-20:
The sysfs gpio interface, that this library uses, supports what you want. All that is needed is to set `direction` to `high`.

Unfortunately, this library does not know about this option and forces `direction` to be either `in` or `out`.

It would be nice to see this fixed.