EnotionZ/gpio #42: Maintenance transfer

rzr, 2018-07-05:
Let me quote @EnotionZ from:
https://github.com/EnotionZ/GpiO/pull/39

"I'm happy to make someone else maintainer if anyone's interested."

I could volunteer for this, and suggest a cooperative maintenance model (for updating npm repo),
but I am new to project ( https://github.com/EnotionZ/GpiO/pull/41 ) 
so if anyone has more experience on this module let us know your plans here,
then @EnotionZ will decide how and when a new version could be released by new maintainers.

My 2c.

EnotionZ, 2018-07-10:
Hey @rzr, thanks for opening this issue, I do appreciate the rigor and consideration. Why don't I make you a collaborator on the project now, and we can decide on when/how to publish the npm module later. I'll stick around to make sure no blatantly harmful code makes it through.

rzr, 2018-07-10:
Ok this is nice from you, myself I have some changes in:
https://github.com/TizenTeam/GpiO

But if you plan to keep the repo active, that would be even better.

Now let's listen for feedback for other potential co-maintainers.

And let's close it once it's updated on npm, if you still plan to "npm publish" it.

rzr, 2018-09-25:
Not related but I just noticed that there is a derived project from @djdeath :-1: 

https://github.com/djdeath/pwm

could it been merged back to main project to provide full abstraction of sysfs for I/O ?
ADC support is also welcome.

djdeath, 2018-09-27:
Hey!
Actually I can't really remember why I created that repo :|
I don't think I'll pursue merging this stuff in your project, but feel free to copy anything you think is interesting.

EnotionZ, 2018-09-27:
@djdeath sounds great, ty!

rzr, 2018-09-27:
@EnotionZ I just pushed trivial changes without review to master branch, I can try to merge @djdeath code unless someone else what to do it.

EnotionZ, 2018-09-28:
@rzr Feel free to pull that in. Can you wait a day or two? I'm cleaning up the file right now to adhere to modern code style standards, just wanna avoid bumping into large conflicts.

EnotionZ, 2018-10-01:
I think this issue is more-or-less resolved and we've created new PR/issues from this discussion. Closing issue.

closed 2018-10-01 by EnotionZ