EnotionZ/gpio #47: Bug in 0.2.7

makeros, 2018-08-01:
Hi. 
For some reason the bug provided here https://github.com/EnotionZ/GpiO/commit/dde7be94be172024fcabbbd8876ec5875c440a01 is available in the 0.2.7 which was released 3 years ago.
Could you enable git tags for while updating the version?

rzr, 2018-08-01:
I think I am granted to tag but not to npm publish, can this be done automatically ?

I imagine @EnotionZ like to release a version once all PR are closed,

makeros, 2018-08-01:
I would be also great to have a CI with some testing running there (i mean something for free like circle ci)
But what about the bug which is no npm in this version?

EnotionZ, 2018-08-01:
I just bumped and published version 0.2.9. I'll look into if there's a way to auto publish when a new version is committed - it must be possible through a git push hook. @rzr  I checked the other day to see if it's possible to assign npm maintainers with publish capabilities without giving them full admin privileges (ie: removing other admins) - apparently this isn't possible. I'm certainly open to it in the future.

Test runner is a great idea. Right now (and since we're testing hardware), one of the tests require manually connecting one pin to another to simulate software trigger on the first pin, which then simulates a hardware trigger on the second. In place of that test, and when https://github.com/EnotionZ/GpiO/pull/40 is merged, we can write a similar test using a custom file to simulate that hardware trigger.

I'm closing this issue and making a new one for automated tests.

closed 2018-08-01 by EnotionZ

rzr, 2018-08-01:
An other option will be to create an org team on npm and relocate the package there.